espressif/esp-box #175: Unresponsive Espressif ESP32-S3-BOX-3

samaro119, 2025-01-02:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [X] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [X] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

Display visible output from the LED indicators or built-in display.

### Actual behavior (suspected bug)

We have an Espressif ESP32-S3-BOX-3 (ID: 5835) that is completely unresponsive and does not show any visible signs of life. The led indicators and screen do not turn on.

When connected to a windows computer the device is accessible and has an active serial port.

The unit was working for a customer for 2 weeks, running ESPHome voice assistant. The device failed suddenly and is now unresponsive.

Here is a list of the troubleshooting steps we have taken:
- Used known working power cables to eliminate variables.
- Checked firmware version to ensure correct firmware
- confirmed that ESP32 is recognised by computer.

ESP Launchpad was used as recommended by the instructions. The quick start page does not work. Using the "Reset Device" option on the Console page results in an error. I have condensed the output to what I believe is relevant but can provide the whole thing if needed.

The intention behind this post is to fix this device or receive confirmation of a fault to pursue a business credit for this component. Are there any further troubleshooting steps that can be done at this stage?

### Error logs or terminal output

```shell
[I][esp_adf:019]: Start codec chip
[D][esp-idf:000]: E (1142) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:104 (i2c_bus_write_bytes):I2C Bus WriteReg Error

[D][esp-idf:000]: E (2144) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:104 (i2c_bus_write_bytes):I2C Bus WriteReg Error

[D][esp-idf:000]: E (3145) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:104 (i2c_bus_write_bytes):I2C Bus WriteReg Error

[D][esp-idf:000]: E (4146) I2C_BUS: components/esp_peripherals/driver/i2c_bus/i2c_bus.c:104 (i2c_bus_write_bytes):I2C Bus WriteReg Error

E (11698) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11698) task_wdt:  - loopTask (CPU 1)
E (11698) task_wdt: Tasks currently running:
E (11698) task_wdt: CPU 0: IDLE0
E (11698) task_wdt: CPU 1: IDLE1
E (11698) task_wdt: Aborting.

abort() was called at PC 0x420a13bc on core 0
```

### Steps to reproduce the behavior

Connect device to power

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

Windows 10

### Shell

PowerShell

espressif2022, 2025-01-03:
Can you provide more information. From the log, the chip is still working, but the i2c master failed to detect slaves.

samaro119, 2025-01-07:
This is the full log from attempting to reset the device using ESP Launchpad

`esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE,Unknown Embedded PSRAM (AP_1v8)
Crystal is 40MHz
MAC: 3c:84:27:05:ad:1c
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Manufacturer: c8
Device: 6018
Detected flash size: 16MB
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c15
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x16c4
load:0x403c9700,len:0xbc0
load:0x403cc700,len:0x2e90
entry 0x403c9950
I (18) boot: ESP-IDF 4.4.7 2nd stage bootloader
I (18) boot: compile time 21:51:49
I (18) boot: Multicore bootloader
I (18) boot: chip revision: v0.2
I (19) boot.esp32s3: Boot SPI Speed : 80MHz
I (19) boot.esp32s3: SPI Mode       : DIO
I (19) boot.esp32s3: SPI Flash Size : 16MB
I (19) boot: Enabling RNG early entropy source...
I (20) boot: Partition Table:
I (20) boot: ## Label            Usage          Type ST Offset   Length
I (20) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (20) boot:  1 phy_init         RF data          01 01 0000b000 00001000
I (21) boot:  2 app0             OTA app          00 10 00010000 007c0000
I (21) boot:  3 app1             OTA app          00 11 007d0000 007c0000
I (22) boot:  4 nvs              WiFi data        01 02 00f90000 0006d000
I (22) boot: End of partition table
I (22) esp_image: segment 0: paddr=00010020 vaddr=3c170020 size=29fcc0h (2751680) map
I (516) esp_image: segment 1: paddr=002afce8 vaddr=3fc9cb10 size=00330h (   816) load
I (517) esp_image: segment 2: paddr=002b0020 vaddr=42000020 size=161f38h (1449784) map
I (777) esp_image: segment 3: paddr=00411f60 vaddr=3fc9ce40 size=04688h ( 18056) load
I (781) esp_image: segment 4: paddr=004165f0 vaddr=40374000 size=18b10h (101136) load
I (814) boot: Loaded app from partition at offset 0x10000
I (814) boot: Disabling RNG early entropy source...
I (814) cpu_start: Multicore app
I (815) opi psram: vendor id : 0x0d (AP)
I (815) opi psram: dev id    : 0x03 (generation 4)
I (815) opi psram: density   : 0x05 (128 Mbit)
I (815) opi psram: good-die  : 0x01 (Pass)
I (815) opi psram: Latency   : 0x01 (Fixed)
I (816) opi psram: VCC       : 0x00 (1.8V)
I (816) opi psram: SRF       : 0x01 (Fast Refresh)
I (816) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (816) opi psram: BurstLen  : 0x01 (32 Byte)
I (817) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (817) opi psram: DriveStrength: 0x00 (1/1)
I (818) MSPI Timing: PSRAM timing tuning index: 6
I (818) spiram: Found 128MBit SPI RAM device
I (818) spiram: SPI RAM mode: sram 80m
I (818) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (819) cpu_start: Pro cpu up.
[... 909 lines cut ...]
[C][wifi:427]:   Local MAC: 3C:84:27:05:AD:1C
[W][wifi_esp32:926]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_CONN
[C][wifi:432]:   SSID: ''
[C][wifi:439]:   BSSID: 00:00:00:00:00:00
[C][wifi:440]:   Hostname: 'esp32-s3-box-05ad1c'
[W][wifi_esp32:937]: esp_wifi_sta_get_ap_info failed: ESP_ERR_WIFI_CONN
[C][wifi:442]:   Signal strength: 0 dB ▂▄▆█
[C][wifi:446]:   Channel: 1
[C][wifi:447]:   Subnet: 0.0.0.0
[C][wifi:448]:   Gateway: 0.0.0.0
[C][wifi:449]:   DNS1: 0.0.0.0
[C][wifi:450]:   DNS2: 0.0.0.0
[C][mdns:116]: mDNS:
[C][mdns:117]:   Hostname: esp32-s3-box-05ad1c
[C][http_request.ota:026]: Over-The-Air updates via HTTP request
[C][esphome.ota:073]: Over-The-Air updates:
[C][esphome.ota:074]:   Address: esp32-s3-box.local:3232
[C][esphome.ota:075]:   Version: 2
[C][safe_mode:018]: Safe Mode:
[C][safe_mode:020]:   Boot considered successful after 60 seconds
[C][safe_mode:021]:   Invoke after 10 boot attempts
[C][safe_mode:023]:   Remain in safe mode for 300 seconds
[W][safe_mode:031]: SAFE MODE IS ACTIVE`

The chip seems to be successfully detected by ESP Launchpad but no visual signs of life are present at any stage from the display or the LED indicators during this process.

espressif2022, 2025-01-07:
I noticed that the issue occurs after some usage and that the firmware has not been reflashed during this time, correct?

It seems that the codec chip initialization failed.
To rule out hardware issues, you can test with the [factory firmware](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-box/launchpad.toml) to check if the recording and speaker functions are working properly.

[image]

samaro119, 2025-01-07:
This is the log from running this test 

`esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE,Unknown Embedded PSRAM (AP_1v8)
Crystal is 40MHz
MAC: 3c:84:27:05:ad:1c
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Manufacturer: c8
Device: 6018
Detected flash size: 16MB
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Compressed 1095184 bytes to 660020...
Writing at 0x0... (2%)
Writing at 0x13893... (4%)
Writing at 0x20aef... (7%)
Writing at 0x26e77... (9%)
Writing at 0x30cde... (12%)
Writing at 0x3890e... (14%)
Writing at 0x46a17... (17%)
Writing at 0x52d80... (19%)
Writing at 0x586fe... (21%)
Writing at 0x5efda... (24%)
Writing at 0x64aae... (26%)
Writing at 0x69e89... (29%)
Writing at 0x6efab... (31%)
Writing at 0x75313... (34%)
Writing at 0x79fc6... (36%)
Writing at 0x7f0a4... (39%)
Writing at 0x83f9a... (41%)
Writing at 0x893b3... (43%)
Writing at 0x8ebdb... (46%)
Writing at 0x93ead... (48%)
Writing at 0x9931a... (51%)
Writing at 0x9e421... (53%)
Writing at 0xa3b72... (56%)
Writing at 0xa9184... (58%)
Writing at 0xaf4f5... (60%)
Writing at 0xb4bc2... (63%)
Writing at 0xbaba5... (65%)
Writing at 0xc00c6... (68%)
Writing at 0xc5e08... (70%)
Writing at 0xce7a6... (73%)
Writing at 0xd6657... (75%)
Writing at 0xdbc96... (78%)
Writing at 0xe09d8... (80%)
Writing at 0xe517f... (82%)
Writing at 0xe9966... (85%)
Writing at 0xedfd6... (87%)
Writing at 0xf27b6... (90%)
Writing at 0xf92c7... (92%)
Writing at 0xfe6dd... (95%)
Writing at 0x103df5... (97%)
Writing at 0x109872... (100%)
Wrote 1095184 bytes (660020 compressed) at 0x0 in 7.679 seconds.
Leaving...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378c18
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
[... 130 lines cut ...]
I (788) main_task: Started on CPU0
I (789) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (789) main_task: Calling app_main()
I (1790) gpio: GPIO[44]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1793) LVGL: Starting LVGL task
I (1794) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2794) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (2794) ili9341: LCD panel create success, version: 1.2.0
W (2914) ili9341: The 36h command has been used and will be overwritten by external initialization sequence
W (2914) ili9341: The 3Ah command has been used and will be overwritten by external initialization sequence
E (5917) ESP-BOX-3: Touch not found
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4201087f
file: "./managed_components/espressif__esp-box-3/esp-box-3.c" line 541
func: bsp_display_indev_init
expression: bsp_touch_new(((void *)0), &tp)

abort() was called at PC 0x4037e4ff on core 0

Backtrace: 0x40375e46:0x3fcc7fc0 0x4037e509:0x3fcc7fe0 0x40384dbe:0x3fcc8000 0x4037e4ff:0x3fcc8070 0x4201087f:0x3fcc80a0 0x420108d9:0x3fcc80f0 0x42008c84:0x3fcc8130 0x42087da7:0x3fcc8310

ELF file SHA256: fe378de4

Rebooting...`

This error repeats and prompts a reboot in a loop from this point until the device is disconnected. During this process, the device does not play any noise or show any output from the screen or LEDs

samaro119, 2025-01-09:
All of the FPC cables looked to be connected correctly.

To be safe all 4 cables have been disconnected and solidly reconnected inside the case. When reassembled and connected to ESP Launchpad the error does not change and the screen remains non-functional.

espressif2022, 2025-01-09:
> All of the FPC cables looked to be connected correctly.
> 
> To be safe all 4 cables have been disconnected and solidly reconnected inside the case. When reassembled and connected to ESP Launchpad the error does not change and the screen remains non-functional.

can you add an [i2c scan function](https://github.com/espressif/esp-iot-solution/blob/86fa25270db8016495169672adbc3f2adfa1bd6f/components/i2c_bus/i2c_bus.c#L125) to check Which slave is not responding？

samaro119, 2025-01-16:
The linked I2C scan function returns 0.

Trying another I2C scan using the Arduino IDE for connection also returns no I2C devices found.

Arduino code used:

`#include <Wire.h>

void setup()
{
  Wire.begin();
  Serial.begin(115200);

  Serial.println("\nI2C Scanner");
}

void loop()
{
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("!");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknown error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("Done.\n");

  delay(5000);          
}`

Output:
"Scanning...
No I2C devices found

Scanning...
No I2C devices found

Scanning...
No I2C devices found"

We have already done substantial testing on this unit. Our intention is to verify that it is faulty so that we may claim a credit for it. At this point can the device be considered as faulty?

samaro119, 2025-01-22:
Hi Team,

Just wanted to check in on this one and see if you had any other troubleshooting steps to recommend or if this device can be considered faulty?

espressif2022, 2025-01-22:
> Hi Team,
> 
> Just wanted to check in on this one and see if you had any other troubleshooting steps to recommend or if this device can be considered faulty?

I'm very sorry about this.  Please contact the customer service for faulty.

ESP-Mars, 2025-03-03:
Hi @samaro119  If you seek after-sales service. Please reach out to the platform where you made the purchase and tell them our that you are already in touch with Espressif on GitHub . The distributor will coordinate with us to process the return or exchange.

ESP-Mars, 2025-04-01:
This issue has remained inactive for an extended period. Closing it now.