espressif/esp-box #177: 关于BOX3跑adf释放SR模型的崩溃问题

PoohWoah, 2025-01-13:
### Checklist

- [X] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [X] Provided a clear description of your suggestion.
- [X] Included any relevant context or examples.

### Issue or Suggestion Description

```
`    if (cfg.sr_handle)
    {
        recorder_sr_destroy(cfg.sr_handle);
        cfg.sr_handle = NULL;
    }

    if (recorder)
    {
        audio_recorder_destroy(recorder);
        recorder = NULL;
    }`
```
    
    释放时遭遇了系统崩溃。
`I (8473) RECORDER_SR: Fetch task destroyed!
I (8484) RECORDER_SR: Feed task destroyed!
I (8506) AFE_SR: afe_se_task quit

I (8506) AFE_SR: afe_mase task destroy finished

I (8507) AFE_SR: exit function: afe_destory

assert failed: remove_free_block tlsf.c:330 (prev && "prev_free field can not be null")

Backtrace: 0x40375f52:0x3c7595e0 0x4037f781:0x3c759600 0x40388bb1:0x3c759620 0x403867e2:0x3c759740 0x40386428:0x3c759760 0x40376a0e:0x3c759780 0x40388be1:0x3c7597a0 0x420804ee:0x3c7597c0 0x42080525:0x3c7597e0 0x4203f6d3:0x3c759800 0x42014a79:0x3c759820 0x42014e31:0x3c759850 0x4200d16e:0x3c759880 0x4200d235:0x3c7598a0 0x40380431:0x3c7598d0
--- 0x40375f52: panic_abort at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/esp_system/panic.c:463
0x4037f781: esp_system_abort at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x40388bb1: __assert_func at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/newlib/assert.c:80
0x403867e2: remove_free_block at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/heap/tlsf/tlsf.c:330 (discriminator 1)
 (inlined by) block_remove at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/heap/tlsf/tlsf.c:380 (discriminator 1)
 (inlined by) block_merge_next at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/heap/tlsf/tlsf.c:486 (discriminator 1)
 (inlined by) tlsf_free at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/heap/tlsf/tlsf.c:1204 (discriminator 1)
0x40386428: multi_heap_free_impl at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/heap/multi_heap.c:235
0x40376a0e: heap_caps_free at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/heap/heap_caps_base.c:70
0x40388be1: free at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/newlib/heap.c:39
0x420804ee: fst_state_free at /home/sunxiangyu/workspace/esp_sr_lib/components/fst/fst.c:147
0x42080525: fst_fst_free at /home/sunxiangyu/workspace/esp_sr_lib/components/fst/fst.c:161
0x4203f6d3: model_destroy at /home/sunxiangyu/workspace/esp_sr_lib/components/multinet/multinet6_quantized.c:318
0x42014a79: recorder_sr_clear at C:/esp-idf/esp-adf/components/audio_recorder/recorder_sr.c:522
0x42014e31: recorder_sr_destroy at C:/esp-idf/esp-adf/components/audio_recorder/recorder_sr.c:643
0x4200d16e: wakeup_pipeline_close at C:/Users/82030/Desktop/Micro_ai/main/app/app_wakeup.c:146
0x4200d235: voice_read_task at C:/Users/82030/Desktop/Micro_ai/main/app/app_wakeup.c:86
0x40380431: vPortTaskWrapper at C:/esp-idf/esp-idf-5.3.1/v5.3.1/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134`
    初始化如下
    
    ```
  recorder_sr_cfg_t recorder_sr_cfg = DEFAULT_RECORDER_SR_CFG();
    recorder_sr_cfg.afe_cfg.memory_alloc_mode = AFE_MEMORY_ALLOC_MORE_PSRAM;
    recorder_sr_cfg.afe_cfg.wakenet_init = WAKENET_ENABLE;
    recorder_sr_cfg.afe_cfg.aec_init = RECORD_HARDWARE_AEC;
    recorder_sr_cfg.afe_cfg.agc_mode = AFE_MN_PEAK_NO_AGC;
    recorder_sr_cfg.afe_cfg.vad_init = VAD_ENBLE;
    recorder_sr_cfg.afe_cfg.vad_mode = VAD_MODE_4;
    recorder_sr_cfg.multinet_init = MULTINET_ENABLE;
    recorder_sr_cfg.mn_language = ESP_MN_CHINESE;

    cfg.read = (recorder_data_read_t)&input_cb_for_afe;
    cfg.sr_handle = recorder_sr_create(&recorder_sr_cfg, &cfg.sr_iface);
    char err[200];
    recorder_sr_reset_speech_cmd(cfg.sr_handle, SPEECH_COMMANDS, err);
    cfg.vad_start = 50;
    cfg.event_cb = rec_engine_cb;
    cfg.vad_off = 600;
    recorder = audio_recorder_create(&cfg);
```

espressif2022, 2025-01-13:
recorder_sr_destroy 是发 event 让线程自己删除的吗？

PoohWoah, 2025-01-14:
> recorder_sr_destroy 是发 event 让线程自己删除的吗？

我发事件？我是直接用的这个api，我看例子里面都是这样删除的呀。

PoohWoah, 2025-01-14:
> recorder_sr_destroy 是发 event 让线程自己删除的吗？

```
    recorder_sr_destroy(cfg.sr_handle);

    audio_recorder_destroy(recorder);
```

这两个

espressif2022, 2025-01-14:
> recorder_sr_destroy

取 ADF 仓库看了下，是的。你这边问题主要和 ADF 关联大，建议可以在对应仓库下提下

closed 2025-04-01 by ESP-Mars