espressif/esp-box #178: ESP32-S3 Box-Lite可以支持吗？

new-soul, 2025-01-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Described the feature in detail and justified the reason for the request.
- [x] Provided specific use cases and examples.

### Feature description

E (543) i2c.master: I2C transaction unexpected nack detected
E (553) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (553) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (563) I2C_If: Fail to write to dev 30
E (573) i2c.master: I2C transaction unexpected nack detected
E (573) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (583) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (593) I2C_If: Fail to write to dev 30
E (593) i2c.master: I2C transaction unexpected nack detected
E (603) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (613) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (613) I2C_If: Fail to write to dev 30
E (623) i2c.master: I2C transaction unexpected nack detected
E (623) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (633) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (643) I2C_If: Fail to write to dev 30
E (643) i2c.master: I2C transaction unexpected nack detected
E (653) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (663) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (663) I2C_If: Fail to write to dev 30
E (673) i2c.master: I2C transaction unexpected nack detected
E (683) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (683) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (693) I2C_If: Fail to write to dev 30
E (693) i2c.master: I2C transaction unexpected nack detected
E (703) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (713) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (723) I2C_If: Fail to write to dev 30
E (723) i2c.master: I2C transaction unexpected nack detected
E (733) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (733) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (743) I2C_If: Fail to write to dev 30
E (753) i2c.master: I2C transaction unexpected nack detected
E (753) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (763) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (773) I2C_If: Fail to write to dev 30
E (773) i2c.master: I2C transaction unexpected nack detected
E (783) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (793) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (793) I2C_If: Fail to write to dev 30
E (803) i2c.master: I2C transaction unexpected nack detected
E (803) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (813) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (823) I2C_If: Fail to write to dev 30
E (823) i2c.master: I2C transaction unexpected nack detected
E (833) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (843) i2c.master: i2c_master_transmit_receive(1097): I2C transaction failed
E (853) I2C_If: Fail to read from dev 30
I (853) ES8311: Work in Slave mode
E (853) i2c.master: I2C transaction unexpected nack detected
E (863) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (873) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (883) I2C_If: Fail to write to dev 30
E (883) i2c.master: I2C transaction unexpected nack detected
E (893) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (893) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (903) I2C_If: Fail to write to dev 30
E (913) i2c.master: I2C transaction unexpected nack detected
E (913) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (923) i2c.master: i2c_master_transmit_receive(1097): I2C transaction failed
E (933) I2C_If: Fail to read from dev 30
E (933) i2c.master: I2C transaction unexpected nack detected
E (943) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (953) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (953) I2C_If: Fail to write to dev 30
E (963) i2c.master: I2C transaction unexpected nack detected
E (973) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (973) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (983) I2C_If: Fail to write to dev 30
E (983) i2c.master: I2C transaction unexpected nack detected
E (993) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1003) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (1013) I2C_If: Fail to write to dev 30
E (1013) i2c.master: I2C transaction unexpected nack detected
E (1023) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1033) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (1033) I2C_If: Fail to write to dev 30
E (1043) i2c.master: I2C transaction unexpected nack detected
E (1043) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1053) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (1063) I2C_If: Fail to write to dev 30
E (1063) ES8311: Open fail

assert failed: BoxAudioCodec::BoxAudioCodec(void*, int, int, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, uint8_t, uint8_t, bool) box_audio_codec.cc:50 (out_codec_if_ != NU

Backtrace: 0x40379fb2:0x3fcb2300 0x403847a9:0x3fcb2320 0x4038d121:0x3fcb2340 0x4200ccbf:0x3fcb2460 0x42019c61:0x3fcb24f0 0x42012f77:0x3fcb2530 0x420166ab:0x3fcb2660 0x421a3e3b:0x3fcb2690 0x40385359:0x3fcb26c0
0x40379fb2: panic_abort at /Users/soul/esp/v5.3/esp-idf/components/esp_system/panic.c:463
0x403847a9: esp_system_abort at /Users/soul/esp/v5.3/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x4038d121: __assert_func at /Users/soul/esp/v5.3/esp-idf/components/newlib/assert.c:80
0x4200ccbf: BoxAudioCodec::BoxAudioCodec(void*, int, int, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, gpio_num_t, unsigned char, unsigned char, bool) at /Users/soul/Documents/Developer/ESP/xiaozhi-esp32/main/audio_codecs/box_audio_codec.cc:50 (discriminator 1)
0x42019c61: EspBox3Board::GetAudioCodec() at /Users/soul/Documents/Developer/ESP/xiaozhi-esp32/main/boards/esp-box-3/esp_box3_board.cc:145
0x42012f77: Application::Start() at /Users/soul/Documents/Developer/ESP/xiaozhi-esp32/main/application.cc:221
0x420166ab: app_main at /Users/soul/Documents/Developer/ESP/xiaozhi-esp32/main/main.cc:28 (discriminator 1)
0x421a3e3b: main_task at /Users/soul/esp/v5.3/esp-idf/components/freertos/app_startup.c:208
0x40385359: vPortTaskWrapper at /Users/soul/esp/v5.3/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

### Use cases

我知道目前ESP32-S3 Box 3可以被支持，不知早期的能否有机会适配。我试了一下，是会报以下错误。应该是I2C部分有问题。
如果是只烧入开发板是可以看到WIFI且被配网，但是不能再下一步了。

espressif2022, 2025-01-20:
可以的，编译时，你可以用 menuconfig 选择 ESP32-S3 Box-Lite，默认是 Box-3

new-soul, 2025-01-20:
> 可以的，编译时，你可以用 menuconfig 选择 ESP32-S3 Box-Lite，默认是 Box-3

看来是要去解决屏的驱动问题了。

espressif2022, 2025-01-20:
> > 可以的，编译时，你可以用 menuconfig 选择 ESP32-S3 Box-Lite，默认是 Box-3
> 
> 看来是要去解决屏的驱动问题了。

不是很理解，你切换后，对应的屏驱动和相应外设都会使用 box-lite 的，不需要自己另外适配

new-soul, 2025-01-20:
> > > 可以的，编译时，你可以用 menuconfig 选择 ESP32-S3 Box-Lite，默认是 Box-3
> > 
> > 
> > 看来是要去解决屏的驱动问题了。
> 
> 不是很理解，你切换后，对应的屏驱动和相应外设都会使用 box-lite 的，不需要自己另外适配

可以会有报错，看头部的。也是蛮奇怪的。

espressif2022, 2025-02-05:
> > > > 可以的，编译时，你可以用 menuconfig 选择 ESP32-S3 Box-Lite，默认是 Box-3
> > > 
> > > 
> > > 看来是要去解决屏的驱动问题了。
> > 
> > 
> > 不是很理解，你切换后，对应的屏驱动和相应外设都会使用 box-lite 的，不需要自己另外适配
> 
> 可是会有报错，看头部的。也是蛮奇怪的。

能贴一下报错的 log？

new-soul, 2025-02-13:
> > > > > 可以的，编译时，你可以用 menuconfig 选择 ESP32-S3 Box-Lite，默认是 Box-3
> > > > 
> > > > 
> > > > 看来是要去解决屏的驱动问题了。
> > > 
> > > 
> > > 不是很理解，你切换后，对应的屏驱动和相应外设都会使用 box-lite 的，不需要自己另外适配
> > 
> > 
> > 可是会有报错，看头部的。也是蛮奇怪的。
> 
> 能贴一下报错的 log？

`E (563) i2c.master: I2C transaction unexpected nack detected
E (563) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (573) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (573) I2C_If: Fail to write to dev 30
E (583) i2c.master: I2C transaction unexpected nack detected
E (593) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (593) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (603) I2C_If: Fail to write to dev 30
E (603) i2c.master: I2C transaction unexpected nack detected
E (613) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (623) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (633) I2C_If: Fail to write to dev 30
E (633) i2c.master: I2C transaction unexpected nack detected
E (643) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (653) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (653) I2C_If: Fail to write to dev 30
E (663) i2c.master: I2C transaction unexpected nack detected
E (663) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (673) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (683) I2C_If: Fail to write to dev 30
E (683) i2c.master: I2C transaction unexpected nack detected
E (693) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (703) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (703) I2C_If: Fail to write to dev 30
E (713) i2c.master: I2C transaction unexpected nack detected
E (713) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (723) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (733) I2C_If: Fail to write to dev 30
E (733) i2c.master: I2C transaction unexpected nack detected
E (743) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (753) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (763) I2C_If: Fail to write to dev 30
E (763) i2c.master: I2C transaction unexpected nack detected
E (773) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (773) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (783) I2C_If: Fail to write to dev 30
E (793) i2c.master: I2C transaction unexpected nack detected
E (793) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (803) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (813) I2C_If: Fail to write to dev 30
E (813) i2c.master: I2C transaction unexpected nack detected
E (823) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (833) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (833) I2C_If: Fail to write to dev 30
E (843) i2c.master: I2C transaction unexpected nack detected
E (843) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (853) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (863) I2C_If: Fail to write to dev 30
E (863) i2c.master: I2C transaction unexpected nack detected
E (873) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (883) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (883) I2C_If: Fail to write to dev 30
E (893) i2c.master: I2C transaction unexpected nack detected
E (903) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (903) i2c.master: i2c_master_transmit_receive(1097): I2C transaction failed
E (913) I2C_If: Fail to read from dev 30
I (923) ES8311: Work in Slave mode
E (923) i2c.master: I2C transaction unexpected nack detected
E (933) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (933) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (943) I2C_If: Fail to write to dev 30
E (953) i2c.master: I2C transaction unexpected nack detected
E (953) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (963) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (973) I2C_If: Fail to write to dev 30
E (973) i2c.master: I2C transaction unexpected nack detected
E (983) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (993) i2c.master: i2c_master_transmit_receive(1097): I2C transaction failed
E (993) I2C_If: Fail to read from dev 30
E (1003) i2c.master: I2C transaction unexpected nack detected
E (1003) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1013) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (1023) I2C_If: Fail to write to dev 30
E (1023) i2c.master: I2C transaction unexpected nack detected
E (1033) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1043) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (1053) I2C_If: Fail to write to dev 30
E (1053) i2c.master: I2C transaction unexpected nack detected
E (1063) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1073) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (1073) I2C_If: Fail to write to dev 30
E (1083) i2c.master: I2C transaction unexpected nack detected
E (1083) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1093) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (1103) I2C_If: Fail to write to dev 30
E (1103) i2c.master: I2C transaction unexpected nack detected
E (1113) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (1123) i2c.master: i2c_master_transmit(1074): I2C transaction failed
E (1133) I2C_If: Fail to write to dev 30
E (1133) ES8311: Open fail`

会不会是音频芯片需要适配，不再是ES8311.
过了年，看到板子支持也是成倍增长，辛苦了！

espressif2022, 2025-02-13:
> `E (563) i2c.master: I2C transaction unexpected nack detected E (563) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed E (573) i2c.master: i2c_master_transmit(1074): I2C [...]

有上电初始化的么，这里看不出是哪个板子？

espressif2022, 2025-02-14:
> > 有上电初始化的么，这里看不出是哪个板子？例如下面是 box 的初始化 log， box-lite，会打印 ESP-BOX-LITE
> > `I (2732) ESP-BOX: Setting LCD backlight: 100%`
> 
> 是box-lite的板子。 有重新上电了。

这样感觉板子是坏了，你加个类似这个 [scan](https://github.com/espressif/esp-box/issues/175#issuecomment-2579202527) 试试 。

ESP-Mars, 2025-04-01:
This issue has remained inactive for an extended period. Closing it now.

closed 2025-04-01 by ESP-Mars