espressif/esp-box #179: ESP32 Box 3 Won't boot after re-flashing

TangoOversway, 2025-01-26:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

I have an ESP32 Box 3. I tried a different firmware and it partially worked. I downloaded the latest Espressif firmware from https://github.com/espressif/esp-box/releases/download/v0.5.0/ESP-BOX_Demo_EN_V0.5.0.bin and flashed it, from my Mac, with the command line `python3 -m esptool --port /dev/cu.usbmodem234401 --chip esp32s3 write_flash 0x0 firmware/ESP-BOX_Demo_EN_V0.5.0.bin`. It keeps rebooting.

Here's the log:

`=00235074 vaddr=3fc9ffI (358) esp_image: segment 1: paddr=00235074 vaddr=3fc9ff00 size=07bbch ( 31676) load
I (364) esp_image: segment 2: paddr=0023cc38 vaddr=40378000 size=033e0h ( 13280) load
I (367) esp_image: segment 3: paddr=00240020 vaddr=42000020 size=14d694h (1365652) map
I (574) esp_image: segment 4: paddr=0038d6bc vaddr=4037b3e0 size=14ac0h ( 84672) load
I (591) esp_image: segment 5: paddr=003a2184 vaddr=50000000 size=00004h (     4) load
I (592) esp_image: segment 6: paddr=003a2190 vaddr=600fe010 size=00004h (     4) load
I (603) boot: Loaded app from partition at offset 0x20000
I (603) boot: Disabling RNG early entropy source...
I (603) cpu_start: Multicore app
I (604) octal_psram: vendor id    : 0x0d (AP)
I (604) octal_psram: dev id       : 0x03 (generation 4)
I (604) octal_psram: density      : 0x05 (128 Mbit)
I (604) octal_psram: good-die     : 0x01 (Pass)
I (605) octal_psram: Latency      : 0x01 (Fixed)
I (605) octal_psram: VCC          : 0x00 (1.8V)
I (605) octal_psram: SRF          : 0x01 (Fast Refresh)
I (605) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (606) octal_psram: BurstLen     : 0x01 (32 Byte)
I (606) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (606) octal_psram: DriveStrength: 0x00 (1/1)
I (607) MSPI Timing: PSRAM timing tuning index: 6
I (607) esp_psram: Found 16MB PSRAM device
I (608) esp_psram: Speed: 80MHz
I (608) cpu_start: Pro cpu up.
I (608) cpu_start: Starting app cpu, entry point is 0x403797bc
I (0) cpu_start: App cpu up.
I (1244) esp_psram: SPI SRAM memory test OK
I (1253) cpu_start: Pro cpu start user code
I (1253) cpu_start: cpu freq: 240000000 Hz
I (1253) cpu_start: Application information:
I (1253) cpu_start: Project name:     factory_demo
I (1254) cpu_start: App version:      v0.3.0-90-gff8376d-dirty
I (1254) cpu_start: Compile time:     Aug 16 2023 17:39:12
I (1254) cpu_start: ELF file SHA256:  5e8aacf9...
I (1254) cpu_start: ESP-IDF:          v5.1-546-ge5199390a3
I (1255) cpu_start: Min chip rev:     v0.0
I (1255) cpu_start: Max chip rev:     v0.99
I (1255) cpu_start: Chip rev:         v0.2
I (1255) heap_init: Initializing. RAM available for dynamic allocation:
I (1255) heap_init: At 3FCAE090 len 0003B680 (237 KiB): DRAM
I (1256) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (1256) heap_init: At 600FE014 len 00001FD4 (7 KiB): RTCRAM
I (1256) esp_psram: Adding pool of 16384K of PSRAM memory to heap allocator
I (1257) spi_flash: detected chip: gd
I (1257) spi_flash: flash io: qio
I (1258) sleep: Configure to isolate all GPIO pins in sleep state
I (1258) sleep: Enable automatic switching of GPIO sleep configuration
I (1259) coexist: coex firmware version: 80b0d89
I (1259) coexist: coexist rom version e7ae62f
I (1259) app_start: Starting scheduler on CPU0
I (0) app_start: Starting scheduler on CPU1
I (1259) main_task: Started on CPU0
I (1260) esp_psram: Reserving pool of 8K of internal memory for DMA/internal allocations
I (1260) main_task: Calling app_main()
I (1261) main: Compile time: Aug 16 2023 17:39:40
I (1449) ESP-BOX: Partition size: total: 2439971, used: 1376986
I (1450) LVGL: Starting LVGL task
I (1451) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
[... 100 lines cut ...]
I (1257) app_start: Starting scheduler on CPU0
I (0) app_start: Starting scheduler on CPU1
I (1257) main_task: Started on CPU0
I (1258) esp_psram: Reserving pool of 8K of internal memory for DMA/internal allocations
I (1258) main_task: Calling app_main()
I (1259) main: Compile time: Aug 16 2023 17:39:40
I (1447) ESP-BOX: Partition size: total: 2439971, used: 1376986
I (1448) LVGL: Starting LVGL task
I (1449) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1449) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (1572) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2
E (1572) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (1573) TT21100: esp_lcd_touch_tt21100_read_data(173): I2C read error!
E (1573) TT21100: esp_lcd_touch_new_i2c_tt21100(103): TT21100 init failed
E (1573) TT21100: Error (0xffffffff)! Touch controller TT21100 initialization failed!
I (1573) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff at 0x4203a70e
file: "./managed_components/espressif__esp-box/esp-box.c" line 389
func: bsp_display_indev_init
expression: bsp_touch_new(((void *)0), &tp)

abort() was called at PC 0x403889a3 on core 0

Backtrace: 0x40379d36:0x3fcb1b40 0x403889ad:0x3fcb1b60 0x4038dbe6:0x3fcb1b80 0x403889a3:0x3fcb1bf0 0x4203a70e:0x3fcb1c20 0x4203a7a5:0x3fcb1c50 0x4203a7d9:0x3fcb1c70 0x42009c92:0x3fcb1cb0 0x4214a10e:0x3fcb1cf0 0x40388f12:0x3fcb1d20
`

espressif2022, 2025-01-27:
[image]

Sorry, I think this firmware doesn't support esp-box-3
You can choose a [firmware ending with -3](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-box/launchpad.toml)

TangoOversway, 2025-01-27:
Yes, that worked!

Where can I download the binary files listed on that page?

Thank you!

Hal

espressif2022, 2025-01-27:
> Yes, that worked!
> 
> Where can I download the binary files listed on that page?
> 
> 
> Thank you!
> 
> 
> 
> Hal
> […](#)

You can first download [launchpad.toml](https://espressif.github.io/esp-box/launchpad.toml) first, and then download according to the firmware name, for example https://espressif.github.io/esp-box/chatgpt_demo-ESP-BOX-1_0_0.bin

TangoOversway, 2025-01-27:
> … <x-msg://5/#>

Just what is launchpad.toml supposed to do? When I click on it, it just opens in my text editor. (I’m on a Mac, in case that makes a difference.)

Thank you.

Hal

espressif2022, 2025-02-05:
@TangoOversway 
launchpad.toml is a configuration file, including the firmware upgrade address and the specific firmware bin name

After you download and open the text, you can see a list of all bin files. Replace chatgpt_demo-ESP-BOX-1_0_0.bin as needed to download the bin file you need.

justjoheinz, 2025-02-26:
@espressif2022  - I made exactly the same mistake. However, now the launchpad does not offer a choice of applications any more, and the links for launchpad.toml etc. do not work anymore. It would be brilliant if you could update the Update Firmware instructions at https://github.com/espressif/esp-box/releases because they say that the 0.5.0 focuses on the S3 Box (which in fact does not seem to be true). 

Please let me know where I can find downloads for the firmware.

justjoheinz, 2025-02-26:
@espressif2022 - I also noticed that the README contains this section:

ESP-Mars, 2025-04-01:
@justjoheinz @TangoOversway  Sorry for late response. The [launchpad.toml](https://espressif.github.io/esp-box/launchpad.toml) file lists all available binary names for the three ESP-BOX series: ESP32-S3-BOX (EOL), ESP32-S3-BOX-Lite (EOL), and ESP32-S3-BOX-3 (Active).
To flash a binary for ESP32-S3-BOX-Lite, simply copy the corresponding file name and replace it in the download link. For example:
- https://espressif.github.io/esp-box/mp3_demo-ESP-BOX-Lite-0_0_1.bin ([Readme](https://github.com/espressif/esp-box/tree/master/examples/mp3_demo) for the example)
- https://espressif.github.io/esp-box/usb_headset-ESP-BOX-Lite-0_1_1.bin ([Readme](https://github.com/espressif/esp-box/tree/master/examples/usb_headset) for the example)
- https://espressif.github.io/esp-box/chatgpt_demo-ESP-BOX-Lite-1_0_0.bin ([Readme](https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo) for the example)

To flash a binary for ESP32-S3-BOX-3, follow the same approach:
- https://espressif.github.io/esp-box/watering_demo-ESP-BOX-3-0_0_1.bin ([Readme](https://github.com/espressif/esp-box/tree/master/examples/watering_demo) for the example)
- https://espressif.github.io/esp-box/factory_demo-ESP-BOX-3-1_2_4 ([Readme](https://github.com/espressif/esp-box/tree/master/examples/factory_demo) for the example)
- https://espressif.github.io/esp-box/chatgpt_demo-ESP-BOX-3-1_0_0.bin ([Readme](https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo) for the example)

Both Mac Os and Windows System work well.