espressif/esp-box #180: Compilation Error in chatgpt_demo Due to Incompatible API Changes in the Latest Version of espressif/button

LeeGartu, 2025-01-29:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The project should compile successfully without any errors.

### Actual behavior (suspected bug)

The project fails to compile, and the following errors are observed.

### Error logs or terminal output

```shell
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c: In function 'lvgl_port_add_navigation_buttons':
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:506:52: error: too few arguments to function 'iot_button_create'
  506 |         buttons_ctx->btn[LVGL_PORT_NAV_BTN_PREV] = iot_button_create(buttons_cfg->button_prev);
      |                                                    ^~~~~~~~~~~~~~~~~
In file included from D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__button/include/iot_button.h:10,
                 from D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/include/esp_lvgl_port.h:31,
                 from D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:17:
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__button/include/button_types.h:53:11: note: declared here
   53 | esp_err_t iot_button_create(const button_config_t *config, const button_driver_t *driver, button_handle_t *ret_button);       
      |           ^~~~~~~~~~~~~~~~~
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:512:52: error: too few arguments to function 'iot_button_create'
  512 |         buttons_ctx->btn[LVGL_PORT_NAV_BTN_NEXT] = iot_button_create(buttons_cfg->button_next);
      |                                                    ^~~~~~~~~~~~~~~~~
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__button/include/button_types.h:53:11: note: declared here
   53 | esp_err_t iot_button_create(const button_config_t *config, const button_driver_t *driver, button_handle_t *ret_button);       
      |           ^~~~~~~~~~~~~~~~~
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:518:53: error: too few arguments to function 'iot_button_create'
  518 |         buttons_ctx->btn[LVGL_PORT_NAV_BTN_ENTER] = iot_button_create(buttons_cfg->button_enter);
      |                                                     ^~~~~~~~~~~~~~~~~
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__button/include/button_types.h:53:11: note: declared here
   53 | esp_err_t iot_button_create(const button_config_t *config, const button_driver_t *driver, button_handle_t *ret_button);       
      |           ^~~~~~~~~~~~~~~~~
In file included from C:/Espressif/frameworks/esp-idf-v5.4/components/esp_system/include/esp_system.h:12,
                 from D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:7:
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:524:88: error: passing argument 3 of 'iot_button_register_cb' from incompatible pointer type [-Wincompatible-pointer-types]
  524 |         ESP_ERROR_CHECK(iot_button_register_cb(buttons_ctx->btn[i], BUTTON_PRESS_DOWN, lvgl_port_btn_down_handler, buttons_ctx));
      |                                                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                                                        |
      |                                                                                        void (*)(void *, void *)
C:/Espressif/frameworks/esp-idf-v5.4/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'   
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__button/include/iot_button.h:107:105: note: expected 'button_event_args_t *' but argument is of type 'void (*)(void *, void *)'
  107 | esp_err_t iot_button_register_cb(button_handle_t btn_handle, button_event_t event, button_event_args_t *event_args, button_cb_t cb, void *usr_data);
      |                                                                                    ~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:524:116: error: 
passing argument 4 of 'iot_button_register_cb' from incompatible pointer type [-Wincompatible-pointer-types]
  524 |         ESP_ERROR_CHECK(iot_button_register_cb(buttons_ctx->btn[i], BUTTON_PRESS_DOWN, lvgl_port_btn_down_handler, buttons_ct 
));
      |                                                                                                                    ^~~~~~~~~~ 
      |                                                                                                                    |
      |                                                                                                                    lvgl_port_nav_btns_ctx_t *
C:/Espressif/frameworks/esp-idf-v5.4/components/esp_common/include/esp_err.h:116:30: note: in definition of macro 'ESP_ERROR_CHECK'   
  116 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__button/include/iot_button.h:107:129: note: expected 'button_cb_t' {aka 'void (*)(void *, void *)'} but argument is of type 'lvgl_port_nav_btns_ctx_t *'
  107 | esp_err_t iot_button_register_cb(button_handle_t btn_handle, button_event_t event, button_event_args_t *event_args, button_cb_t cb, void *usr_data);
      |                                                                                                                     ~~~~~~~~~~~~^~
D:/Work/example/esp-box/examples/chatgpt_demo/factory_nvs/managed_components/espressif__esp_lvgl_port/esp_lvgl_port.c:524:25: error: too few arguments to function 'iot_button_register_cb'
[... 44 lines cut ...]
Just compiling the `chatgpt_demo/factory_nvs` project.

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win11

### Shell

ZSH

### Additional context

Adding espressif/button: "^3.5.0" at the end of the idf_component.yml file in chatgpt_demo/factory_nvs can resolve the issue. However, the root cause of this problem is that espressif/esp-box-3 depends on a button version >= 2.5.

espressif2022, 2025-02-05:
Thanks for the reminder. I checked button v4.0+ and it is indeed not compatible. Other demos also have some compatibility issues. I will fix them together.

LeeGartu, 2025-02-05:
If you have completely resolved this issue, could you please explain the approach you took to fix it? I could learn about dependency-related knowledge in the ESP-IDF framework from your explanation.

espressif2022, 2025-02-05:
> If you have completely resolved this issue, could you please explain the approach you took to fix it? I could learn about dependency-related knowledge in the ESP-IDF framework from your explanation.

I also wrote the button version in idf_component.yml in the components/bsp directory

button 4.0 here is a breaking change, and didn’t consider compatibility with the old interface.

You can refer to [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html)

spacefrogg, 2025-02-08:
Thanks, this is working well!

sebastianohl, 2025-02-08:
just for everyone not wanting to read the docs.
add the following dependency to the idf_component.yml in the main folder of your example:
  espressif/button: "<4.0.0"

at least for the lv_demos example this works perfectly

espressif2022, 2025-02-14:
Updated now, please pull the newest version, thanks.

closed 2025-02-21 by LeeGartu