espressif/esp-box #181: Buffer for OpenAI API Key is not big enough

alikian, 2025-01-30:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

The Api  Key Size buffer define in ChatGPT demo is 64 bytes while OpenAI API Key is well over 160 byes
#define KEY_SIZE 64

Suggest to increase it to 256

### Actual behavior (suspected bug)

When testing ChatGPT demo this message appears
"code": invalid_api_key!

### Error logs or terminal output

```shell
app_main: start_openai(91): [audioTranscription]: invalid response
```

### Steps to reproduce the behavior

- Upload chatgpt demo to esp32 s3 box
- put value in config.ini include wifi ssid/password and OpenAI API Key
- reboot, you will see
"code": invalid_api_key!

### Project release version

latest

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

MacOS

### Operating system version

MacOS

### Shell

Bash

espressif2022, 2025-02-05:
I calculated it, it should be 51 bytes, why you will get 160 byes?

alikian, 2025-02-09:
They changed the key size, please test it your self try to get a new project API key or user API Key
- User API Key now is 96 characters
- Admin API Key is 134 characters
- Project API Key is 164 characters

alikian, 2025-02-23:
Just, follow up did you apply changes, thanks

espressif2022, 2025-03-04:
> Just, follow up did you apply changes, thanks

I also increased the key length in the commit of the github workflow action. Please confirm. Thank you very much.

ESP-Mars, 2025-04-01:
The current issue has been resolved, so we are closing this issue.