espressif/esp-box #183: esp32s3-box 运行显示touch not found

zajzhuaijun, 2025-03-05:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4200c022
--- 0x4200c022: bsp_init_main_button at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../managed_components/espressif__esp-box-3/esp-box-3.c:635 (discriminator 1)

file: "../managed_components/espressif__esp-box-3/esp-box-3.c" line 635
func: bsp_init_main_button
expression: bsp_touch_new(((void *)0), &tp)

abort() was called at PC 0x40381ecf on core 0
--- 0x40381ecf: _esp_error_check_failed at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/esp_system/esp_err.c:49

Backtrace: 0x40379e56:0x3fca41f0 0x40381ed9:0x3fca4210 0x403897f5:0x3fca4230 0x40381ecf:0x3fca42a0 0x4200c022:0x3fca42d0 0x420151e2:0x3fca42f0 0x4200c05d:0x3fca4320 0x4200c2be:0x3fca4340 0x4200c443:0x3fca4370 0x4200b17c:0x3fca4390 0x420384f7:0x3fca4410 0x40382a1d:0x3fca4440
--- 0x40379e56: panic_abort at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/esp_system/panic.c:463
0x40381ed9: esp_system_abort at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x403897f5: abort at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/newlib/abort.c:38
0x40381ecf: _esp_error_check_failed at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/esp_system/esp_err.c:49
0x4200c022: bsp_init_main_button at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../managed_components/espressif__esp-box-3/esp-box-3.c:635 (discriminator 1)
0x420151e2: iot_button_create at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../managed_components/espressif__button/iot_button.c:471
0x4200c05d: bsp_iot_button_create at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../managed_components/espressif__esp-box-3/esp-box-3.c:652
0x4200c2be: bsp_btn_init at /home/zhuaijun/Desktop/esp/esp-box/components/bsp/src/boards/esp32_bsp_board.c:92
0x4200c443: bsp_board_init at /home/zhuaijun/Desktop/esp/esp-box/components/bsp/src/boards/esp32_bsp_board.c:247
0x4200b17c: app_main at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../main/main.c:64
0x420384f7: main_task at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/freertos/app_startup.c:208
0x40382a1d: vPortTaskWrapper at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134, 本身显示屏不带触摸功能

### Actual behavior (suspected bug)

W (691) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (701) sleep: Configure to isolate all GPIO pins in sleep state
I (708) sleep: Enable automatic switching of GPIO sleep configuration
I (715) main_task: Started on CPU0
I (725) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (728) main_task: Calling app_main()
I (732) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (743) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1751) gpio: GPIO[48]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1751) ili9341: LCD panel create success, version: 1.2.0
W (1875) ili9341: The 36h command has been used and will be overwritten by external initialization sequence
W (1875) ili9341: The 3Ah command has been used and will be overwritten by external initialization sequence
I (1885) ESP-BOX-3: Setting LCD backlight: 100%
I (1897) button: IoT Button Version: 3.5.0
I (1897) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1904) button: IoT Button Version: 3.5.0
I (1909) gpio: GPIO[1]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1918) button: IoT Button Version: 3.5.0
E (4923) ESP-BOX-3: Touch not found
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4200c022
--- 0x4200c022: bsp_init_main_button at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../managed_components/espressif__esp-box-3/esp-box-3.c:635 (discriminator 1)

file: "../managed_components/espressif__esp-box-3/esp-box-3.c" line 635
func: bsp_init_main_button
expression: bsp_touch_new(((void *)0), &tp)

abort() was called at PC 0x40381ecf on core 0
--- 0x40381ecf: _esp_error_check_failed at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/esp_system/esp_err.c:49

Backtrace: 0x40379e56:0x3fca41f0 0x40381ed9:0x3fca4210 0x403897f5:0x3fca4230 0x40381ecf:0x3fca42a0 0x4200c022:0x3fca42d0 0x420151e2:0x3fca42f0 0x4200c05d:0x3fca4320 0x4200c2be:0x3fca4340 0x4200c443:0x3fca4370 0x4200b17c:0x3fca4390 0x420384f7:0x3fca4410 0x40382a1d:0x3fca4440
--- 0x40379e56: panic_abort at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/esp_system/panic.c:463
0x40381ed9: esp_system_abort at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x403897f5: abort at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/newlib/abort.c:38
0x40381ecf: _esp_error_check_failed at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/esp_system/esp_err.c:49
0x4200c022: bsp_init_main_button at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../managed_components/espressif__esp-box-3/esp-box-3.c:635 (discriminator 1)
0x420151e2: iot_button_create at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../managed_components/espressif__button/iot_button.c:471
0x4200c05d: bsp_iot_button_create at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../managed_components/espressif__esp-box-3/esp-box-3.c:652
0x4200c2be: bsp_btn_init at /home/zhuaijun/Desktop/esp/esp-box/components/bsp/src/boards/esp32_bsp_board.c:92
0x4200c443: bsp_board_init at /home/zhuaijun/Desktop/esp/esp-box/components/bsp/src/boards/esp32_bsp_board.c:247
0x4200b17c: app_main at /home/zhuaijun/Desktop/esp/esp-box/examples/usb_headset/build/../main/main.c:64
0x420384f7: main_task at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/freertos/app_startup.c:208
0x40382a1d: vPortTaskWrapper at /home/zhuaijun/Desktop/esp/esp-adf/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

### Error logs or terminal output

```shell

```

### Steps to reproduce the behavior

编译运行examples/factory_demo这个项目

### Project release version

master

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

esp-adf a2ab6cefd8f9618a0470fd3f549c913fdf7ca1b8

### Shell

ZSH

espressif2022, 2025-03-05:
Hello, your device is esp-box-3 or esp-box ?

zajzhuaijun, 2025-03-06:
my device is  esp32s3-box

espressif2022, 2025-03-06:
> my device is esp32s3-box

Hello, you should reselect HMI_Board in menuconfig

zajzhuaijun, 2025-03-06:
which one should I select？

espressif2022, 2025-03-06:
> which one should I select？

[image]

zajzhuaijun, 2025-03-06:
yes，I did choose this one，but，still run failed with aboved error

espressif2022, 2025-03-06:
[image]
Please notice the log, like here.From your log, the compiled BSP is incorrect.

zajzhuaijun, 2025-03-06:
ok , I try again.thanke you!!

zajzhuaijun, 2025-03-06:
that's true，it‘s ok now。but when I test factory_demo,it failed。

[image]

espressif2022, 2025-03-06:
> that's true，it‘s ok now。but when I test factory_demo,it failed。
> 
> [image]

I can't get any valid information. Please try to re-burn the firmware or try [this fw](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-box/launchpad.toml)

zajzhuaijun, 2025-03-06:
ok.

zajzhuaijun, 2025-03-06:
when I use esp32s3-box test speech_recogintion under adf-framework, idf.py menuconfig choose esp32s3-box,open VOICE2FILE,compile ok, but run failed.
(8211) AUDIO_RECORDER: RECORDER_CMD_TRIGGER_START
I (8211) wwe_example: rec_engine_cb - REC_EVENT_WAKEUP_START
I (8221) wwe_example: wakeup: vol 0.000000, mod idx 0, word idx 0
I (8241) main_task: Returned from app_main()
I (8251) CODEC_ELEMENT_HELPER: The element is 0x3c0e382c. The reserve data 2 is 0x0.
I (8931) SDCARD: Using 1-line SD mode, base path=/sdcard
I (8931) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (8931) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (8941) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (8981) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
HINT: Please reboot the board and then try again
E (8981) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
HINT: Please verify if there is an SD card inserted into the SD slot. Then, try rebooting the board.
I (8981) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (9001) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (9001) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (9011) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (9021) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (9041) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (9041) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (9051) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (9061) SDCARD: Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.
E (9071) PERIPH_SDCARD: mount sdcard error!
I (9931) SDCARD: Using 1-line SD mode, base path=/sdcard
I (9931) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (9941) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (9951) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (10001) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
HINT: Please reboot the board and then try again
E (10001) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
HINT: Please verify if there is an SD card inserted into the SD slot. Then, try rebooting the board.
I (10001) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (10021) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (10021) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (10031) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (10041) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (10061) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (10061) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (10071) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (10081) SDCARD: Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.
E (10091) PERIPH_SDCARD: mount sdcard error!
I (10931) SDCARD: Using 1-line SD mode, base path=/sdcard
I (10931) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (10931) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (10941) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (10991) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
HINT: Please reboot the board and then try again
E (10991) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
HINT: Please verify if there is an SD card inserted into the SD slot. Then, try rebooting the board.
I (10991) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (11001) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (11011) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (11021) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (11041) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (11041) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (11051) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (11071) gpio: GPIO[0]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (11071) SDCARD: Failed to initialize the card (263). Make sure SD card lines have pull-up resistors in place.
E (11081) PERIPH_SDCARD: mount sdcard error!
I (11931) SDCARD: Using 1-line SD mode, base path=/sdcard
I (11931) gpio: GPIO[13]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (11941) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (11951) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (12021) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
[... cut ...]
I (12931) gpio: GPIO[11]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (12941) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
E (12991) sdmmc_common: sdmmc_init_ocr: send_op_cond (1) returned 0x107
HINT: Please reboot the board and then try again
E (12991) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
HINT: Please verify if there is an SD card inserted into the SD slot. Then, try rebooting the board.by the way, I insert a sd-card into card slot on board

espressif2022, 2025-03-06:
This is an adf demo? I think you'd better open another issue in the corresponding repository.

ESP-Mars, 2025-04-01:
Current issue has been resolved, so it is now closed. For the ADF-related problem reported by the user, we recommend opening a new issue.

closed 2025-04-01 by ESP-Mars