espressif/esp-box #185: Request for Example: ESP32-S3-BOX with AEC

sylvieyi, 2025-04-10:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hi, thank you for your great work on this project!

I am currently developing an audio application on the ESP32-S3-BOX, and I would like to implement Acoustic Echo Cancellation (AEC). I have read the documentation and tried the AEC-related functions provided in the esp-sr and algorithm_stream modules, but the echo is still clearly audible in my setup. It seems the current setup is not working well on the ESP32-S3-BOX.

It would be very helpful if you could provide:
	•	An example project that runs on ESP32-S3-BOX with AEC properly configured
	•	Recommended parameters and pipeline setup for better AEC performance

Thanks in advance!

ESP-Mars, 2025-04-10:
Just to confirm—your device is the ESP32-S3-BOX, not the ESP32-S3-BOX-3, right?

sylvieyi, 2025-04-10:
> Just to confirm—your device is the ESP32-S3-BOX, not the ESP32-S3-BOX-3, right?

Apologies for the confusion — I was referring to BOX-3, not BOX. The description has been updated.

espressif2022, 2025-04-15:
> > Just to confirm—your device is the ESP32-S3-BOX, not the ESP32-S3-BOX-3, right?
> 
> Apologies for the confusion — I was referring to BOX-3, not BOX. The description has been updated.

You can refer to [this](https://github.com/espressif/esp-adf/blob/master/components/audio_stream/aec_stream.c) first.