espressif/esp-box #186: ESP32-S3-BOX-3 Network set to factory default

Fbo06, 2025-04-19:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate.
- [x] Provided a clear description of your suggestion.
- [x] Included any relevant context or examples.

### Issue or Suggestion Description

Hello, can you please tell me how to restore to the factory the network connection, i try to maintain at the boot time the power button for 5 to 10 secondes but each time my old connection is restored. i want to do this because on my phone i can add a new device on the ESP-BOX app. thank you for your advice

ESP-Mars, 2025-04-21:
Please long press the Boot button for 5 seconds to clear the network information and restore the device to factory settings.
More user guide for the factory please refer to https://github.com/espressif/esp-box/blob/master/docs/getting_started.md#voice-command-customization

Fbo06, 2025-04-21:
Thank you ESP-Mars, this is what the documentation, which i read and try before writing this post, said but nothing work, the network does not go to factory setting but each time the connection is well restored to my wifi. this fact do that i cannot found my ESP32-S3-BOX-3 on the new phone APP when i try to add a new device on the phone app. so the question is how to set the network to the factory setting ?

Fbo06, 2025-05-24:
@ESP-Mars Thank you for your help, in fact there is another one button i push. now it's ok, another time sorry to bother you. thank you for your help