espressif/esp-box #188: 升级过程中意外中断后无法再次升级

yuanfangcn, 2025-04-27:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

期望能正常烧录固件

### Actual behavior (suspected bug)

idf.py -p COM3 flash 升级过程中（约烧录了10%多）因意外中断，后续重新 按boot，再按reset后释放，即尝试再次升级，但是出现：无法停留在 升级状态（waiting for download），而是接着自动和反复重启，从而导致 无法再次升级

### Error logs or terminal output

```shell
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x20 (DOWNLOAD(USB/UART0))
waiting for download //此处没有稳定等待升级，而是若干秒后自动进行了 重启，导致无法再次升级
EESSESESSSESSEESESESSSESESESPESEESESESESPESESESPESPESPESP-ESP-ESP-----RESP-ESP--R--R-ESP-----R-RESP-R-ROESP-RO-ROM-RO-ROMESP-ROM:-RO-ROMESP-ROM-ROM:ESP-ROM:esp-ROM:esp-ROM:es-ROM:es-ROM:es-ROM:es-ROM:espESP-ROM:espESP-ROM:es-ROM:esp-ROM:esp-ROM:esp-ROM:esp3-ROM:esp-ROM:esp-ROM:esp3-ROM:esp-ROM:esp32-ROM:esp3-ROM:esp32s3-ROM:esp32s3-ROM:esp32s-ROM:esp32s3--ROM:esp3-ROM:esp32ESP-ROM:esp3-ROM:esp32ESP-ROM:esp3-ROM:esp3-ROM:esp32s-ROM:esp32-ROM:esp32s-ROM:esp32s-ROM:esp32sESP-ROM:esp32sESP-ROM:esp32s-ROM:esp32s-ROM:esp32s-ROM:esp32s-ROM:esp32s3-ROM:esp32s3--ROM:esp32s3-ROM:esp32s3-ESP-ROM:esp32s3-ESP-ROM:esp32s3-2-ROM:esp32s3-2-ROM:esp32s3-2-ROM:esp32s3-2-ROM:esp32s3-2-ROM:esp32s3-2-ROM:esp32s3-20-ROM:esp32s3-2-ROM:esp32s3-20-ROM:esp32s3-202-ROM:esp32s3-2021-ROMESP-ROM:esp32s3-2021032732s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (27) boot: compile time Apr 21 2025 18:15:14
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 002a0000
I (78) boot:  3 storage          Unknown data     01 82 002b0000 00100000
I (85) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c1d0020 size=7b940h (506176) map
E (184) esp_image: invalid segment length 0xffffffff
E (185) boot: Factory app partition is not bootable
E (185) boot: No bootable app partitions in the pa-ROM:esp32s3-2021032-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
Build:Mar -ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5ESPES-ESP-ESP--ESP----ESP--R-ESP---ESESSESESSESESESPESESESESSESPESESESESESESPESESPESPESP-ESP-ESP----R--ESP-------R--R-RESP-R-R-R-ROESP-ROESP-ROESP-RO-ROESP-R-ROM-R-ROMESP-ROM:-ROM-ROM-ROM:-ROM:ESP-ROM:-ROM:e-ROM:e-ROM:e-ROM:e-ROM:e-ROM:e-ROM:e-ROM:e-ROM:es-ROM:e-ROM:es-ROM:es-ROM:es-ROM:esp3ESP-ROM:esp-ROM:esp32ESP-ROM:esp3-ROM:esp32-ROM:esp32-ROM:esp32-ROM:esp32-ROM:esp32-ROM:esp32s-ROM:esp32s-ROM:esp32s3-ROM:esp32s-ROM:esp32s3-ROM:esp32s3ESP-ROM:esp32s3ESP-ROM:esp32s3--ROM:esp32s3-ROM:esp32s3-ROM:esp32s3-2-ROM:esp32s3--ROM:esp32s3--ROM:esp32s3-20ESP-ROM:esp32s3-2-ROM:esp32s3-20-ROM:esp32s3-202-ROM:esp32s3-2021-ROM:esp32s3-202-ROM:esp32s3-2021-ROM:esp32s3-20210-ROM:esp32s3-20210-ROM:esp32s3-20210-ROM:esp32s3-20210327-ROM:esp32s3-20210327ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (27) boot: compile time Apr 21 2025 18:15:14
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot:-ROM:esp32s3--ROM:esp32s3-202-ROM:esp32s3-2021032-ROM:esp32s3-20210327-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4-dirty 2nd stage bootloader
I (27) boot: compile time Apr 21 2025 18:15:14
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.2
I (30) boot: efuse block revision: v1.3
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 002a0000
I (78) boot:  3 storage          Unknown data     01 82 002b0000 00100000
I (85) boot: End of partition table
```

### Steps to reproduce the behavior

1.esp32s3-wroom-1-n4r2 因某次升级过程中意外中断升级(中途断电)
2.尝试再次升级，按boot后再按reset和释放，
3.无法稳定的停在等待升级的模式，而是自动重启，导致无法再次升级

### Project release version

idf5.4

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Windows

### Operating system version

win10

### Shell

PowerShell

espressif2022, 2025-04-27:
rst:0x1 (POWERON),boot:0x0 (DOWNLOAD(USB/UART0))
waiting for download
上述 log 是自动打印，不是手动按下 boot + reset 打印的吗？

无法稳定停留下 booot 阶段，本地是否启动的 idf.py flash 命令？

看你是自己设计了PCB 板？麻烦硬件检查下 rst 脚波形

yuanfangcn, 2025-04-27:
是手动按下 boot + reset 打印的。
是设计过的PCB板。
你的意思是 查下RST波形线路，即使是意外中断烧录，也不会影响 再次操作烧录是嘛

espressif2022, 2025-04-27:
> 是手动按下 boot + reset 打印的。 是设计过的PCB板。 你的意思是 查下RST波形线路，即使是意外中断烧录，也不会影响 再次操作烧录是嘛

是的，boot 阶段没有外围电路影响是不会重启的，如果你手上有 dev-kit 可以验证下

closed 2025-04-29 by yuanfangcn