espressif/esp-box #197: 红外学习码部分报错问题 esp-s3-box-3

ypcgoal, 2025-08-04:
### Checklist

- [x] Checked the issue tracker for similar issues to ensure this is not a duplicate
- [x] Read the documentation to confirm the issue is not addressed there and your configuration is set correctly
- [x] Tested with the latest version to ensure the issue hasn't been fixed

### How often does this bug occurs?

always

### Expected behavior

我在验证红外学习部分的代码时出现了学习成功，但是日志报错的问题，最终红外还是不能控制空调的开关，我的idf版本是5.5

### Actual behavior (suspected bug)

目前我学习的是格力的空调开关机，虽然界面上提示学习成功，并显示了可以控制空调的按钮，但是并没有成功开关机空调。

### Error logs or terminal output

```shell
I (35060) ui_sensor_monitor: IR Learn ready
I (35185) ui_sensor_monitor: tips_info: Learning failed, retry!
I (36689) ui_sensor_monitor: tips_info: Press the button
I (43070) ui_sensor_monitor: IR Learn step:[1][1]
I (43243) ui_sensor_monitor: IR Learn step:[1][2]
I (43288) ui_sensor_monitor: tips_info: Press button again
I (45179) ui_sensor_monitor: IR Learn step:[2][1]
I (45349) ui_sensor_monitor: IR Learn step:[2][2]
I (45388) ui_sensor_monitor: tips_info: First learning successful
I (46585) ui_sensor_monitor: tips_info: Learn again, ready!
I (47804) ui_sensor_monitor: IR Learn step:[3][1]
I (47976) ui_sensor_monitor: IR Learn step:[3][2]
I (48089) ui_sensor_monitor: tips_info: Press button again
I (49183) ui_sensor_monitor: IR Learn step:[4][1]
I (49353) ui_sensor_monitor: IR Learn step:[4][2]
I (49853) ir learn: list:1-2
I (49853) ir learn: list:2-2
I (49853) ir learn: list:3-2
I (49853) ir learn: list:4-2
E (49854) ir learn: ./managed_components/espressif__ir_learn/ir_learn.c:415 (ir_learn_check_duration):add cmd duration error
E (49854) ir learn: ir_learn_check_valid(491): symbol add failed
I (49854) ir learn: list:1-2
I (49854) ir learn: list:2-2
I (49854) ui_sensor_monitor: IR Learn on ok
I (49854) ir learn: list:1-2
I (49855) ir learn: list:2-2
I (49855) ui_sensor_monitor: IR Learn off ok
I (51716) ui_sensor_monitor: IR Learn exit
```

### Steps to reproduce the behavior

红外学习期间如果开关按键的时间间隔控制不好，很容易学习失败，需要多次尝试才能正确学习到红外码

### Project release version

idf版本5.5

### System architecture

Intel/AMD 64-bit (modern PC, older Mac)

### Operating system

Linux

### Operating system version

ubuntu24.04

### Shell

Bash

espressif2022, 2025-08-04:
这边应该是出现了，某次电平相差过大的时序。 每次都是失败的吗？
单独试下 https://github.com/espressif/esp-iot-solution/tree/master/components/ir/ir_learn/test_apps

ypcgoal, 2025-08-05:
> 这边应该是出现了，某次电平相差过大的时序。 每次都是失败的吗？ 单独试下 https://github.com/espressif/esp-iot-solution/tree/master/components/ir/ir_learn/test_apps

使用了你说的这个测试样例，学习时还是报错：
Here's the test menu, pick your combo:
(1)	"IR learn and send test" [IR][IOT][Manual]
(2)	"IR keep learn test" [IR][IOT][Manual]
(3)	"IR self-receiver test" [IR][IOT][Auto]

Enter test for running.
1
Running IR learn and send test...
I (19703) ir learn: IR learn Version: 1.0.0
I (19705) ir_learn_test: IR Learn ready
I (26087) ir_learn_test: IR Learn step:[1][1]
I (26149) ir_learn_test: IR Learn step:[1][2]
I (26323) ir_learn_test: IR Learn step:[1][3]
I (30853) ir_learn_test: IR Learn step:[2][1]
I (31025) ir_learn_test: IR Learn step:[2][2]
I (33802) ir_learn_test: IR Learn step:[3][1]
I (33976) ir_learn_test: IR Learn step:[3][2]
I (36095) ir_learn_test: IR Learn step:[4][1]
I (36203) ir_learn_test: IR Learn step:[4][2]
I (36267) ir_learn_test: IR Learn step:[4][3]
I (36768) ir learn: list:1-3
I (36768) ir learn: list:2-2
E (36768) ir learn: ir_learn_check_valid(468): cmd num mismatch
E (36768) ir_learn_test: IR Learn failed, retry
I (49065) ir_learn_test: IR Learn step:[1][1]
I (49239) ir_learn_test: IR Learn step:[1][2]
I (51043) ir_learn_test: IR Learn step:[2][1]
I (51103) ir_learn_test: IR Learn step:[2][2]
I (51275) ir_learn_test: IR Learn step:[2][3]
I (52539) ir_learn_test: IR Learn step:[3][1]
I (52713) ir_learn_test: IR Learn step:[3][2]
I (53910) ir_learn_test: IR Learn step:[4][1]
I (54082) ir_learn_test: IR Learn step:[4][2]
I (54582) ir learn: list:1-2
I (54582) ir learn: list:2-3
E (54582) ir learn: ir_learn_check_valid(468): cmd num mismatch
E (54582) ir_learn_test: IR Learn failed, retry
I (56621) ir_learn_test: IR Learn step:[1][1]
I (56682) ir_learn_test: IR Learn step:[1][2]
I (56857) ir_learn_test: IR Learn step:[1][3]
I (58069) ir_learn_test: IR Learn step:[2][1]
I (58241) ir_learn_test: IR Learn step:[2][2]
I (59309) ir_learn_test: IR Learn step:[3][1]
I (59371) ir_learn_test: IR Learn step:[3][2]
I (59545) ir_learn_test: IR Learn step:[3][3]
I (60513) ir_learn_test: IR Learn step:[4][1]
I (60685) ir_learn_test: IR Learn step:[4][2]
I (61186) ir learn: list:1-3
I (61186) ir learn: list:2-2
E (61186) ir learn: ir_learn_check_valid(468): cmd num mismatch
E (61186) ir_learn_test: IR Learn failed, retry
claude分析说，是学习了4轮，虽然是奇数次学习开机命令，偶数次学习关机命令，但是最后验证的时候却是混在一起验证的，所以导致验证会失败。

espressif2022, 2025-08-13:
```
const static rmt_receive_config_t ir_learn_rmt_rx_cfg = {
    .signal_range_min_ns = 1000,
    .signal_range_max_ns = 10 * 1000 * 1000,
};
```
ir_learn 的组件可以先本地改下` signal_range_max_ns ` 时间